rotorflight/rotorflight #10: OSD and VTX table not work in Matek F722 PX/HD

RipperGOT, 2022-05-08:
What the title indicates. In the configurator the OSD or VTX tab does not appear, I have tried to activate it via CLI but it does not work either.
[image]

rotorflight, 2022-05-08:
Video support has been disabled during the development, simply because there has been zero interest to it, and none of the developers has a system to test it either. I would be happy to enable it if someone (may you?) could help with testing. Come join the Discord chat if you want to get involved.

RipperGOT, 2022-05-09:
OK understood! I'd be happy to help where possible! But I have no programming knowledge. The FC where I am testing Rotorflight is the Matek F722 PX (FrSky Pixel OSD). I do not have any HD video system. All my quads and airplanes are with analog video systems. Even the heli I'm testing it on!
But if I can help with anything, it would be a pleasure.