rotorflight/rotorflight #15: Ethos..

robthomson, 2023-07-24:
So. whats it going to take to get the lua scripts ported to Ethos?

Do I need to send somebody a radio as motivation?

robthomson, 2023-07-25:
Ok.. so a little bit of chatting with Frsky and they are going to put one of their developers onto the task of converting the Betaflight Lua scripts to Ethos.

I suspect this will to all intents be a massive 'help' for this project as essentially a 'fork' ?

Shout if you need a link-up

rotorflight, 2023-07-25:
Indeed the biggest issue is to find someone with enough FrSky gear to develop and test it, plus the necessary skills to do it. AFAIK none of RF devs is using Ethos based FrSky gear.

It will help a lot if we had Ethos LUA scripts for BF. It would be straightforward to convert the RF scripts to the same format.

James-T1, 2023-07-25:
This was done a year and a half ago:

https://github.com/James-T1/rotorflight-lua-ethos

robthomson, 2023-07-26:
Wow Wow Wow..

Its 'clunky' but functionally works!  Just tested on latest version of ethos.

So on the plus side.. frsky is in theory getting BF done... and for now we have this one working.

I will have a play and see if I can make the UI of this one a bit cleaner - mostly it could maybe do with some re-sizing and touch screen enabling.  No idea if I will achieve it but will try anyway!

egonl, 2023-07-26:
I'm the Rotorflight dev maintaining the LUA scripts, using a Q X7 on OpenTX + R-XSR and a TX16S on Edge + ELRS EP1/2.

I'm aware of James's work, but since I don't have an Ethos transmitter I can't test any modifications. If FrSky were able to provide one, I would happily maintain Rotorflight LUA scipts for Ethos as well.

robthomson, 2023-07-27:
Hey.. I have passed request over to them.  Will find out soonish.

robthomson, 2023-07-27:
> Rob if you are not in Rotorflight Discord yet, please join and make some noise. Most of the development discussion happens there.
> 
> https://discord.com/invite/FyfMF4RwSA

Joined!

robthomson, 2023-07-30:
> I'm the Rotorflight dev maintaining the LUA scripts, using a Q X7 on OpenTX + R-XSR and a TX16S on Edge + ELRS EP1/2.
>
> I'm aware of James's work, but since I don't have an Ethos transmitter I can't [...]

Hey..   Ping on on discord please?  Not sut who you are on there.

rotorflight, 2023-11-27:
Ethos support has been added. Closing this!

closed 2023-11-27 by rotorflight