rotorflight/rotorflight #27: Heli vibration level displayed on the transmitter via telemetry

heliphysics, 2025-07-24:
To be able to adjust the blade tracking of a helicopter ideally, it would be helpful to have the vibration level of the helicopter displayed on the transmitter via telemetry. 

This would allow corrections to be made directly on the flying field in order to find the ideal setting. Up to now, this has only been possible via a blackbox and is therefore very cumbersome, which means that fine-tuning is very rarely carried out or only when obviously necessary. 
A smooth-running system protects the components and improves control. Such a function would also be very useful for detecting mechanical problems on the helicopter at an early stage.

Would it be possible to calculate a value from the sensor data that describes the vibrations of the helicopter? 
The aim is not to determine an absolute value, but to determine whether the vibration level improves or deteriorates after a correction on the helicopter.

robthomson, 2025-07-24:
This is not overly complex to do on ethos - but would depend very much on ability to log this effecvively over time.

black box allows you to drill down to very fine details.  prob with telem is the gaps may be too large to be usefull.   Something to ponder

heliphysics, 2025-07-25:
I think this feature is very important, it will find many friends and further enhance RF.
I have already heard from many RF users that they would like such a function.

heliphysics, 2025-07-25:
The perfect adjustment of the blade tracking could then look like this:

1. stationary hovering, reading and memorizing the vibration value on the Tx
2. land and slightly extend the linkage of the marked rotor blade
3. hover again and read off the value
4. if it is lower than before, extend the same boom further, otherwise shorten it by twice the value
5. repeat until the lowest vibration level is reached

In this way, a maximum smooth-running helicopter can be achieved within a few minutes.